ahkab/ahkab #29: Does not work with Ipython

orena1, 2015-04-27:
Hi.
I have a problem with the package and ipython.
As soon as I dd
import ahkab
from ahkab import circuit, printing, devices
 
First thing that I get is:
W: Locale appears not set! please export LANG="en_US.UTF-8" or equivalent, 
W: or ahkab's unicode support is broken.

and then when I use the up arrow key in the keyboard I get this:
```python
In [4]: ^[[A
```
If I use the down arrow key I get:
```python
In [5]: ^[[B
```
I use linux
Linux  3.14.14-gentoo
```python
In [2]: for i in numpy, scipy, matplotlib:
   ...:         print i.__version__
   ...:     
1.9.2
0.14.0
1.4.3

Python 2.7.9 |Anaconda 2.1.0 (64-bit)| (default, Mar  9 2015, 16:20:48) 
Type "copyright", "credits" or "license" for more information.

IPython 2.1.0 -- An enhanced Interactive Python.

```

locale -a
gives me -
```
C
en_US
en_US.iso88591
en_US.utf8
POSIX
```

and doing  export LANG="en_US" change noting.

Thanks.

ggventurini, 2015-04-27:
Hi Oren,

Thanks for the bug report.

I can reproduce the arrow issue, I'll be looking into that. Maybe you could try the notebook as a temporary work-around?

For the unicode issue, can you `export LANG="en_US.UTF-8"` instead of `"export LANG="en_US"` and report back any change?

Thanks!

ggventurini, 2015-04-27:
Also, from preliminary testing the ipython console running in Python 3 seems to be unaffected.

```
Python 3.3.5 |Anaconda 1.9.1 (x86_64)| (default, Sep  2 2014, 13:57:31) 
Type "copyright", "credits" or "license" for more information.

IPython 3.1.0 -- An enhanced Interactive Python.
Anaconda is brought to you by Continuum Analytics.
Please check out: http://continuum.io/thanks and https://binstar.org
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import ahkab
# presses up arrow
In [2]: import ahkab

In [3]: 
Do you really want to exit ([y]/n)? y
```

ggventurini, 2015-04-27:
I believe that should sort out your issue alright.

The fix will be included in the next release, if nothing new shows up.

As that might be a few weeks from now, if you wish, you can run the code in the master branch, or run ipython and ahkab in Python3.

closed 2015-04-27 by ggventurini

ggventurini, 2015-04-27:
Thanks a lot for the bug report! +1

pammacdotnet, 2018-11-22:
I still get UnicodeEncodeError with Python 3.6.6

commit 976766dc (2015-04-27): BUGFIX: prevent breaking readline on ipython, fixes #29