ahkab/ahkab #30: Ahkab floods /tmp/ directory

RJTK, 2015-05-18:
I am using Ahkab to model circuits for to test some optimization strategies.  So, I am running an Ahkab simulation hundreds or thousands of times very quickly.  This quickly leads to my program crashing due to having too many open files.  It turns out that every time I run an analysis, Ahkab creates a file in /tmp/.

I've looked through the source and it seems that if you set "outfile" to None in new_ac it will create a temp file for the output.

from ahkab.py, in each new_ function.
```python
    if guess is None:
        guess = options.dc_use_guess
    if outfile is None or outfile == 'stdout':
        if options.cli:
            outfile = 'stdout'
        else:
            ofi, outfile = tempfile.mkstemp(suffix='.op')
            _of.append(ofi)  # keep the file open until quitting
```
The fd is stored in the _of list, but I haven't found anywhere they get closed.

I also am unsure about why it's necessary to write to this temp file in the first place?

As far as I can tell, there is no obvious way to disable writing to these temp files.

ggventurini, 2015-05-18:
Hi,

I can see what the problem is.

Simulations may generate a lot of data. The approach we take is that we
write the data to disk, and read it when the user accesses. If no output
file is specified, then a temporary one is created to hold the data. Many
OS's delete the temporary files when they are closed, so it's never closed,
and that indeed is a bug.

As a work-around, you may specify an output file. After the simulation is
run and you have processed your results, you may run a new one with the
same output file, (which will get overwritten).

Hope that helps for the time being. I will look into fixing the too many
open files issue soon.

How many files did you have open and what OS do you use? I'd have imagine
that number to be pretty high.

Cheers,

RJTK, 2015-05-18:
I'm running Linux Mint 17 and it has about 1500 open files when it crashes.

ggventurini, 2015-05-18:
The latest master fixes the temporary files being left open. We still rely on the OS to clear /tmp from time to time, ie we still write files there.

I think this partially addresses your issue. Please let me know if it prevents the crashes you reported.

RJTK, 2015-05-18:
It doesn't crash any more.  It does still flood the /tmp/ directory, but that isn't really a bug.  I think the best thing to do is for the user to just specify a hidden file for it to write to.

commit 31b9f9bc (2015-05-18): Do not leave open tmp files around. Partially fix #30
commit 651f2795 (2015-05-22): Use atexit to purge temporary files. fix #30
closed 2015-05-22 by ggventurini