ahkab/ahkab #31: UnicodeEncodeError on win7 64bit OS

chinghua0731, 2015-06-21:
I tried to run the example on my windows OS. But i got the error message below.
(code from :https://raw.githubusercontent.com/ahkab/ahkab/master/doc/misc/script.py)

My Environment is Win7 64bit OS + Python 3.4.2 .
How could I fix this problem , thanks.

Anthony

Error message from Ipython
=================
* Butterworth Example circuit
R1 n1 n2 600
L1 n2 n3 0.01524
C1 n3 0 1.1937e-07
L2 n3 n4 0.06186
C2 n4 0 1.5512e-07
R2 n4 0 1200
V1 n1 0 type=vdc value=5 vac=1 type=pulse v1=0 v2=1 td=5e-07 per=2 tr=1e-12 tf=1e-12 pw=1
W: Locale appears not set! please export LANG="en_US.UTF-8" or equivalent, 
W: or ahkab's unicode support is broken.
---------------------------------------------------------------------------
UnicodeEncodeError                        Traceback (most recent call last)
<ipython-input-1-7b6d7d5389c7> in <module>()
     23 tran_analysis = ahkab.new_tran(tstart=0, tstop=1.2e-3, tstep=1e-6, x0=None)
     24 
---> 25 r = ahkab.run(mycircuit, an_list=[op_analysis, ac_analysis, tran_analysis])
     26 
     27 import pylab

C:\Python34\lib\site-packages\ahkab\ahkab.py in run(circ, an_list)
    657                                    (an_type.upper(), an_item['x0']))
    658             an_item['x0'] = None
--> 659         r = analysis[an_type](circ, **an_item)
    660         results.update({an_type: r})
    661         if an_type == 'op':

C:\Python34\lib\site-packages\ahkab\dc_analysis.py in op_analysis(circ, x0, guess, outfile, verbose)
    682 
    683     if opsolution and outfile != 'stdout' and outfile is not None:
--> 684         opsolution.write_to_file()
    685     if opsolution and (verbose > 2 or outfile == 'stdout') and options.cli:
    686         opsolution.write_to_file(filename='stdout')

C:\Python34\lib\site-packages\ahkab\results.py in write_to_file(self, filename)
    466         fp.write("========================\n\n")
    467         for k in sorted(self._op_info.keys()):
--> 468             fp.write(printing.table(self._op_info[k], headers=self._op_keys[k][0]))
    469             fp.write('\n\n')
    470         #for opi in self.op_info:

UnicodeEncodeError: 'cp950' codec can't encode character '\u03d5' in position 13: illegal multibyte sequence

ggventurini, 2015-06-22:
Hi,

Thanks for the feedback. I will look into this issue.

I have updated the script file to take into account unrelated changes in the library. Would you like to download it again and run it directly with Python? Do you get the same error? with both 3 and 2?

I do not have easy access to a WIndows PC, I will be looking into this issue but it may take a bit.

Thanks.

Cheers,

GV

chinghua0731, 2015-06-22:
Hi GV,

I download the script file from below link, and run it again.But I got same error message .
https://raw.githubusercontent.com/ahkab/ahkab/master/doc/misc/script.py

I also noted another error message show: " Error: The system can not find the path specified "
if i run the script in command line. Is this the same situation mentioned in Issue#30 ? 
(https://github.com/ahkab/ahkab/issues/30)
You mentioned that many OS's delete the temporary files when they are closed.

if so, could you show me how to specify an output file in python code ?

Anthony

ggventurini, 2015-06-22:
Hi,

Did you try to run the latest master? I have merged a commit
specifically for your case, I'd like to know if it makes things
better.

The error message you refer to is a harmless error that happens on
windows when the program checks for matplotlib. you can safely ignore
that.

Cheers,

GV

chinghua0731, 2015-06-23:
Hi GV,

Yes , i tried the latest master but it doesn't make anything change. I still got the same error message.

 I take a look into the code today and have some finding. I found the define functions of _get_elements_op() in results.py (line 339) return some character that can't be identified in windows . But not a problem in Mac .

In Windows
[image]

In Mac
[image]

ggventurini, 2015-06-23:
Hi,

Thanks for your update!

I think it's the Ohm unit symbol. I will be trying to use the same encoding
as you do and see if i can reproduce that.

Unicode handling sometimes is complicated.

Cheers

GV

ggventurini, 2015-06-24:
Hi Anthony,

Thanks for your detailed reply. I think I understood what goes on. 

If I am right, Python3 does not open files as utf-8 on systems where utf-8 is not the standard. Now, starting with 69b10630d78107bfb180756a4504aaf855c358cb we enforce that files are always opened as utf-8.

That should fix your problem, I believe. Can you please give the latest master a try and let me know?

If that works I'll shortly put out a bugfix release.

Thanks!

Cheers, GV

chinghua0731, 2015-06-24:
Hi GV,

Good news, it works!! I could run the script normally on Windows.

I am glad to have such a great package to run the circuit simulation in python. 
It really help me a lot : ) and thanks for your effort on this issue. 

Anthony

ggventurini, 2015-06-24:
Hi Anthony, 

I am pleased to hear the good news. Thanks for your report, and for the nice words. I am pleased it is useful.

Cheers,

GV

closed 2015-06-24 by ggventurini