ahkab/ahkab #33: Support for Frequency-Dependent Components?

cyounkins, 2015-07-08:
What would it take to add frequency-dependent resistors and inductors? I'm looking at modeling Coilcraft inductors in SPICE (http://www.coilcraft.com/modelsltpice.cfm), and they suggest using a Laplace directive to create the component. I don't think Ahkab currently supports this.

Would this require modifying the AC analysis routine to be more complicated? I think you could no longer just do the invert-multiply per step.

ggventurini, 2015-07-08:
I find that to be very interesting and, since it is user-requested, it has great importance in my eyes.

I also have a long list of pile up work that I am trying to get some free time to properly address.

That said, I will look into this soon, but it will have to be after the 0.18 release which is due in the next days.

Cheers!

cyounkins, 2015-07-16:
I wanted to write here and say that while I think this would be a great feature to have, I am unlikely to use it myself as my project has gone on without that part of the simulation. I'll let you decide if you want to pursue this or not - feel free to close if not!

itdaniher, 2017-02-02:
@cyounkins Could you elaborate upon your use case a little more? I'm triaging a bit and trying to decide if adding laplace netlist directives like http://www.ee.ryerson.ca/~courses/ele322/public_html/example2/pspice_example2.html would be worth exploring, or if the platform would benefit more from additional symbolic demos, as I think the symbolic analysis / transfer function derivation and manipulation functionality in Ahkab is underutilised and extremely powerful. The alternative to `.LAPLACE` and friends is directly manipulating transfer functions in Python, akin to what I do in my undercommented notebook, ie [here](https://github.com/itdaniher/ahkab-notebook/blob/master/compensators.py) and [here](https://github.com/itdaniher/ahkab-notebook/blob/master/sk.py).

itdaniher, 2017-02-02:
@cyounkins more demos at https://ahkab.readthedocs.io/en/latest/examples/Symbolic-simulation.html

cyounkins, 2017-02-03:
Thanks for the comment, but my use of ahkab was over a year ago. Frankly I don't remember exactly what I was trying to do.

closed 2017-02-03 by cyounkins