ahkab/ahkab #40: Plain text component values

endolith, 2016-08-25:
Hi, I made this for parsing plain text schematic values:

https://gist.github.com/endolith/1a74477fc1659cd6bf9d138e09c72672

```
value_parser('100 MΩ')
Out[194]: (100000000.0, u'ohm')

value_parser('2M2')
Out[195]: (2200000.0, None)

value_parser('4n7')
Out[196]: (4.700000000000001e-09, None)

value_parser('4.7 nH')
Out[197]: (4.700000000000001e-09, u'henry')
```

So you could make entering components more readable and catch mistakes more easily, too:

```
cir.add_resistor('R1', 'n1', 'n2', '50 ohm')
cir.add_inductor('L1', 'n2', 'n3', '245.894m')
cir.add_capacitor('C1', 'n3', 'n4', '103.013 nF')
cir.add_inductor('L2', 'n4', cir.gnd, '98.3652 μH')
cir.add_vsource('V1', 'n1', cir.gnd, dc_value='0 V'., ac_value='1 V')
```

Is this something you'd be interested in incorporating?

For instance in `Circuit.add_resistor`, it would be something like this:

```
value, unit = value_parser(value)
if unit not in {None, 'ohm'}:
    raise Exception('Unit doesn't match component type')
if value == 0:
    raise CircuitError("ZERO-valued resistors are not allowed.")
```
or put the exception inside the function:

```
value = value_parser_and_checker(value, should_be='ohm')
if value == 0:
    raise CircuitError("ZERO-valued resistors are not allowed.")
```

So entering units would be optional, but would help catch errors of changing the wrong value, etc.  I think most SPICE implementations ignore units.

itdaniher, 2017-02-01:
Hey there! Somehow this originally fell threw the cracks in my inbox, but the improvement looks great and I'd love to figure out how to get it merged!

itdaniher, 2017-02-03:
Thinking about this further:

I like 

```
value = value_parser_and_checker(value, should_be='ohm')
if value is None:
    raise CircuitTypeError("Circuit components need to have nonzero values)
```

This way, I think we have a utility function to add, which you already wrote, and hooks to add to the `__init__` methods of the relevant classes, or, less generally, in the `Circuit.add_foo` methods for each circuit element.

Personally, I'd move even the existing `if value == 0` check into the `__init__` calls for each circuit element, as they're only implemented for a few select components, and it might be worth adding sanity checks that keep someone from trying to simulate the transient response of a circuit with a 1 megafarad capacitor and a 1 megaohm resistor, or other equally awkward unit-ed component combinations.

itdaniher, 2017-02-03:
We already have more infrastructure for this than I realised, as G took the initiative to work towards SPICE syntax compatibility!

http://ahkab.readthedocs.io/en/latest/netlist_parser.html?highlight=convert_units#ahkab.netlist_parser.convert_units

Found just reading netlist_parser thinking about unicode units.....

endolith, 2017-02-04:
SPICE annoyingly equates `M` with "milli", when it means "mega" everywhere else.  Also what is mil = 25.4x1e-6 for?  PCB dimensions?