ahkab/ahkab #42: Custom nonlinear components

gehrer, 2016-12-24:
Hello!

I am trying to implement a resistor with a voltage drop following the formula U=const.*I^2. It is needed for the simulation of a transient process.
I made a class like the description in the device file, but I when I try run it there is only the error message Missing attribute 'get_output_ports' from the circuit file. I do not understand what the circuit file exactly wants. Are there any documentations or examples for the use of nonlinear elements?

Best regards

itdaniher, 2016-12-25:
Hi @gehrer, can you gist the code that produces this error? I'm not an expert in ahkab internals, but as close as there is to a maintainer at this point. If you share your code, I'll try and take a look.

Good luck!

gehrer, 2016-12-25:
Hi,
here is the link to the gist:
https://gist.github.com/gehrer/12c8228b245b21eeaa0a0071127c5e1e

itdaniher, 2016-12-25:
@gehrer thanks, I'll take a look ~tomorrow. Merry Christmas!

itdaniher, 2016-12-27:
@gehrer take a look at https://github.com/ahkab/ahkab/blob/master/ahkab/diode.py#L87, https://github.com/ahkab/ahkab/blob/master/ahkab/diode.py#L125

Looks like it's just part of the API that the above code missed.

itdaniher, 2016-12-27:
try https://gist.github.com/itdaniher/b595075f30c9db440b0409c9b16f835d

looks like nonlinear elements requires a different API than for linear, I tried to make changes as appropriate.

Look at the diode.py source, esp ```class diode```.

regards!
Ian

gehrer, 2016-12-28:
Thank you. Works well now.
Another question: According to the documentation the step size is controlled and probably gets adjusted during the transient simulation. Is there an easy way to force a step size? I sometimes have the problem that an error "Step size too small" occurs, independently of the start step size.

itdaniher, 2016-12-29:
wrt step size:

transient_analysis has the following parameter:

> use_step_control: the LTE will be calculated and the step adjusted. default: True

pretty sure setting it to false will use a constant step size.

you can also set a smaller value for options.hmin, the smallest allowed step size.

Regards,
Ian

itdaniher, 2016-12-30:
Closing this issue as the primary question was addressed - feel free to open another issue if needed!

Regards,
Ian

closed 2016-12-30 by itdaniher