ahkab/ahkab #57: Issue with ankab

unnir, 2017-03-03:
Hi all,

I'm trying to start ahkab on **mac, python 3.5**, and getting these error messages:

```
W: Locale appears not set! please export LANG="en_US.UTF-8" or equivalent, 
W: or ahkab's unicode support is broken.
/python3.5/site-packages/ahkab/plotting.py:235: MatplotlibDeprecationWarning: pyplot.hold is deprecated.
    Future behavior will be consistent with the long-time default:
    plot commands add elements without first clearing the
    Axes and/or Figure.
  pylab.hold(True)
/python3.5/site-packages/matplotlib/__init__.py:917: UserWarning: axes.hold is deprecated. Please remove it from your matplotlibrc and/or style files.
  warnings.warn(self.msg_depr_set % key)
/anaconda/lib/python3.5/site-packages/matplotlib/rcsetup.py:152: UserWarning: axes.hold is deprecated, will be removed in 3.0
  warnings.warn("axes.hold is deprecated, will be removed in 3.0")
/anaconda/lib/python3.5/site-packages/ahkab/plotting.py:248: MatplotlibDeprecationWarning: pyplot.hold is deprecated.
    Future behavior will be consistent with the long-time default:
    plot commands add elements without first clearing the
    Axes and/or Figure.
  pylab.hold(False)
```

Could please help me, I can't understand why I have to change the LANG for starting the script.

itdaniher, 2017-03-03:
Hi there! 

This looks like a standardish issue with Python 3 - it's generally best practices to set LOCALE/LANG to tell the interpreter your native charset. There's probably a more subtle issue here that relates to how OSX sets up its terminal environment, but if setting LANG addressed your issues then... Keep doing that. I don't have an OSX install to test with, unfortunately.

The other warnings are merely reflective of changes in the matplotlib api I haven't gotten around to sussing out.

Closing for now, please re-open if setting LANG manually didn't do the trick.

closed 2017-03-03 by itdaniher

itdaniher, 2017-03-03:
FYI all Linux environments I have access to set LANG for terminal environments at init. Definitely an OSX oddity.

unnir, 2017-03-06:
sorry for this follow-up question, do you have any idea why I'm facing this error?

`/python3.5/site-packages/ahkab/plotting.py:235: MatplotlibDeprecationWarning: pyplot.hold is deprecated.
    Future behavior will be consistent with the long-time default:
    plot commands add elements without first clearing the
    Axes and/or Figure.
  pylab.hold(True)
/python3.5/site-packages/matplotlib/__init__.py:917: UserWarning: axes.hold is deprecated. Please remove it from your matplotlibrc and/or style files.
  warnings.warn(self.msg_depr_set % key)
/anaconda/lib/python3.5/site-packages/matplotlib/rcsetup.py:152: UserWarning: axes.hold is deprecated, will be removed in 3.0
  warnings.warn("axes.hold is deprecated, will be removed in 3.0")
/anaconda/lib/python3.5/site-packages/ahkab/plotting.py:248: MatplotlibDeprecationWarning: pyplot.hold is deprecated.
    Future behavior will be consistent with the long-time default:
    plot commands add elements without first clearing the
    Axes and/or Figure.
  pylab.hold(False)`

I'm trying to start the script from the first page:
```
from ahkab import new_ac, run
from ahkab.circuit import Circuit
from ahkab.plotting import plot_results # calls matplotlib for you
import numpy as np

cir = Circuit('Butterworth 1kHz band-pass filter')
cir.add_vsource('V1', 'n1', cir.gnd, dc_value=0., ac_value=1.)
cir.add_resistor('R1', 'n1', 'n2', 50.)
cir.add_inductor('L1', 'n2', 'n3', 0.245894)
cir.add_capacitor('C1', 'n3', 'n4', 1.03013e-07)
cir.add_inductor('L2', 'n4', cir.gnd, 9.83652e-05)
cir.add_capacitor('C2', 'n4', cir.gnd, 0.000257513)
cir.add_inductor('L3', 'n4', 'n5', 0.795775)
cir.add_capacitor('C3', 'n5', 'n6', 3.1831e-08)
cir.add_inductor('L4', 'n6', cir.gnd, 9.83652e-05)
cir.add_capacitor('C4', 'n6', cir.gnd, 0.000257513)
cir.add_capacitor('C5', 'n7', 'n8', 1.03013e-07)
cir.add_inductor('L5', 'n6', 'n7', 0.245894)
cir.add_resistor('R2', 'n8', cir.gnd, 50.)

ac1 = new_ac(.97e3, 1.03e3, 1e2, x0=None)

res = run(cir, ac1)

plot_results('5th order 1kHz Butterworth filter', [('|Vn8|',"")], res['ac'],
             outfilename='bpf_transfer_fn.png')
```

itdaniher, 2017-03-07:
Yes - these can be safely ignored. More `matplotlib` API depreciation warnings that I haven't attended. Just got back from vacation, will probably clean up some of these this week. 

These shouldn't impact the functionality of Ahkab at all, but are annoying enough that I'll fix them shortly.

Stay tuned!

unnir, 2017-03-07:
thank you, have a nice vacation!