ahkab/ahkab #58: Tran example : 'op_solution' object has no attribute 'asmatrix'

unnir, 2017-03-09:
I'm trying to set-up the [tran example]( https://ahkab.readthedocs.io/en/latest/examples/Transient-Example.html) from the docs.

however, I'm getting a weird error message. 

```
  x0 = res['op'].asmatrix()
AttributeError: 'op_solution' object has no attribute 'asmatrix'
```

itdaniher, 2017-03-09:
you want `asarray` not `asmatrix`

https://github.com/ahkab/ahkab/blob/master/ahkab/results.py#L322

itdaniher, 2017-03-09:
I'll have to fix the docs. Not sure when that method was renamed.

closed 2017-03-09 by itdaniher

unnir, 2017-03-09:
> I'll have to fix the docs

yep, also I will try to help...

regarding the tran example, I've changed to asarray and got a new error:

```
  x0[l1i, 0] += -1e-9
IndexError: index 12 is out of bounds for axis 0 with size 9
```

itdaniher, 2017-03-10:
hmmm, try ```x0[-1,0] += -1e-9```

unnir, 2017-03-10:
> hmmm, try x0[-1,0] += -1e-9

fixed.

Matplotlib again crushed the script, it's the same issue as we discussed it here [issue57](https://github.com/ahkab/ahkab/issues/57 ). I remember you have a vacation now, so just write here/or make an update then you will fix it.