ahkab/ahkab #59: voltage pulse source for a MOS inverter + questions

unnir, 2017-04-02:
Hi all,

I'm working on simple CMOS inverter now, and cannot figure out how to add a pulse source for the inverter. 

Currently, I'm using a sin source instead of the pulse:

```python
inv = ahkab.Circuit('MOS Inverter')
#add models
inv.add_model('ekv', 'nmos', dict(TYPE='n', VTO=0.4, KP=10e-6))
inv.add_model('ekv', 'pmos', dict(TYPE='n', VTO=-0.4, KP=1e-6))

inv.add_vsource('vdd', n1='dd', n2=inv.gnd, dc_value=3.3)

#add MOSFETs

inv.add_mos('m1', nd='output1', ng='in1', ns='dd', nb='dd',
            model_label='pmos', w=600e-6, l=100e-9)
inv.add_mos('m2', nd='output1', ng='in1', ns=inv.gnd, nb=inv.gnd,
            model_label='nmos', w=600e-6, l=100e-9)
#voltage sin source function
mys = lambda t: 1 if not t else math.sin(math.pi*1e4*t)/(math.pi*1e4*t)

inv.add_vsource('V1', 'in1', inv.gnd, 1, function=mys)

op = ahkab.new_op()
print(op)
res = ahkab.run(inv, op)
print(res)
x0 = res['op'].asarray()
x0[-1, 0] += -1e-9

tran = ahkab.new_tran(tstart=0., tstop=1e-6, tstep=.01e-9, method='trap',
                       x0=x0)
res = ahkab.run(inv, tran)['tran']
```

And then I'm trying to plot my input signal:

Question:

- Are there ways to improve the performance of computations for tran simulations?
Because it's unbelievable slow, f.i. about 7 minutes for the tran simulation with tstop=1e-6.
- How to add a pulse source to the circuit? I found in documentations you can create your [own time function](http://ahkab.readthedocs.io/en/latest/time_functions.html
), but it seams to me it does not really work now, anyway I need some hints how to make it.
- Also, I would like to add my inverter to the example folder, is it ok guys?

itdaniher, 2017-04-06:
I'd be happy to have your inverter added.

As for performance, pypy has the potential to improve it, but this isn't trivial until numpy support is first class.

I'll look into pulse inputs!

Regards,
Ian

unnir, 2017-04-06:
Thank you Ian!

itdaniher, 2017-04-06:
NP, keen to help. Want to open a PR with your inverter demo?

unnir, 2017-04-08:
sure, give me please some time, 

but it would be better to PR the inverter with the pulse source :)