ahkab/ahkab #63: RL circuit vs. LR circuit: different output

moredroplets, 2017-10-04:
Thanks for your time reading this thread. I hope you could give me some suggestions regarding Ahkab library. 

I am an engineer specialised in inkjet printhead design. I am trying to use Python Ahkab library to simulate a printhead design where the equivalent flow resistance, inertance, and capacitance are not constant, but variables depending on flow development profile in microfluidic channels. Ahkab is a perfect library for such a job.

However, I found something really hard to comprehend (py file is attached). For a simple RL circuit, the time constant should be L/R. However, the simulated result is not so. Also, by switching the positions of the R and the L components, the result is different for the current running through the inductor.  By comparing them with the result from LTSpice model, both ones are not completely correct. 

Could you give me some suggestions as to what might have gone wrong with my coding or the library itself? 

Many thanks, 

Xi 

#code: 

import numpy as np 
import scipy as sp 
import pylab as plt
import tabulate 
import matplotlib 
import ahkab
from ahkab import new_ac, run
from ahkab.circuit import Circuit
from ahkab.plotting import plot_results # calls matplotlib for you
from ahkab import circuit, printing, time_functions
#########################################################################
cir = circuit.Circuit('Cylinder')
#V_const = time_functions.pulse(v1=0, v2=DeltaPressure, td=1e-12, tr=1e-12, pw=1E-3, tf=1e-12, per=1E-1)
V_const = time_functions.pulse(v1=0, v2=DeltaPressure, td=0., tr=0., pw=1E-3, tf=0., per=1E-1)
#cir.add_vsource('Vconst', 'n1', cir.gnd, dc_value=0., ac_value=0., function=V_const)
#it's the same result using this one. 
#cir.add_vsource('Vconst', 'n1', cir.gnd, dc_value=30000, ac_value=0., function=V_const)
#using DC_Value still the same 
cir.add_vsource('Vconst', 'n1', cir.gnd, dc_value=30000, ac_value=0)
#cir.add_resistor('RnozSS', 'n1', 'n2', 1.45069e+12)      
#cir.add_inductor('LnozSS', 'n2', cir.gnd, 1.14282e+07)  
#######change the positions of Lnoz and Rnoz, the results are different #################
cir.add_inductor('LnozSS', 'n1', 'n2', 1.14282e+07)  
cir.add_resistor('RnozSS', 'n2', cir.gnd, 1.45069e+12)      
#########################################################################################
print(cir)
T = 20e-5    
deltaT = 20e-9  
#initialise IC
ic = {'V(n1)':0, #30000, 
      'V(n2)':0., 
      'I(LnozSS)':0., 
     }
cylinder_ic = ahkab.new_x0(cir, ic)  
tran_analysis = ahkab.new_tran(tstart=0, 
                               tstop=T, 
                               tstep=deltaT, 
                               x0=cylinder_ic, 
                               #method='GEAR6',
                               #use_step_control=False)                               
                               use_step_control=True)
  
#############################################################################
r = ahkab.run(cir, an_list=tran_analysis)
#print(r['tran']['T'].size)
#print(r['tran']['VN1'][-1])d
#print(r['tran']['VN2'][-1])
#print(r['tran']['I(LnozSS)'][-1])
fig = plt.figure()
plt.title(cir.title + " - TRAN Simulation")
plt.plot(r['tran']['T']*1E+6, r['tran']['I(LnozSS)']*1E+9, label="SPICE")
#plt.plot(temp_timeList*1E+6, temp_flowRateList*1E+9, label="White's Formula")
plt.legend()
plt.grid(True)
plt.xlim([0,T*1E+6])
plt.ylim([0,r['tran']['I(LnozSS)'][-1]*1.05*1E+9])
plt.ylabel('Flow rate [nA]')
plt.xlabel('Time [us]')
fig.savefig('tran_plot.png')
plt.show() 

(This is the LTSpice circuit and output for the current through the inductor)
[image]
[image]

(Below two images are for the current through the inductor using Python Ahkab library. However, the positions of R and L are switched in the circuit. Results are not the same. The bottom image gives the correct final current value but the time constant is wrong.) 
[image]
[image]

itdaniher, 2017-10-04:
edited with block quotes and a definition of `DeltaPressure = 1`

the crux of the issue is the size of the inductor (and probably to a lesser extent, the resistor) - 1E7 henries is uh probably the size of your average house, with 1E3 henries as the approximate inductance of a 3 gigawatt mains transformer...

From > https://en.wikipedia.org/wiki/Orders_of_magnitude_(inductance)

moredroplets, 2017-10-07:
@itdaniher  I need some time to figure out how to scale down the equivalent inductance and resistance for a microfluidic device. As such a system is almost always highly nonlinear, I think it would be very hard to do so.  Is that possible to make the library F64 internally if it's F32? Thanks for your help.

itdaniher, 2017-10-07:
After checking, it's all using Python's default float type, with no specified width as one might expect were it using numpy datatypes internally. This means it's already F64, which suggests the issue is not as simple as using a datatype with insufficient precision.

moredroplets, 2017-10-10:
One way to get around this is to change the unit system from "kg, meter, second" to something else, e.g., "g, um, second", so that the fluidic equivalent inductance, resistance, and capacitance values are more in line with what Ahkab can handle. This should work without changing anything in this library.

itdaniher, 2017-10-10:
Sounds like something of a fix. Hopefully enough to get let you make forward progress in your work!