ahkab/ahkab #70: AttributeError: module 'pylab' has no attribute 'hold'

raczben, 2019-06-12:
I get this error during run the "hello world" example:

```
from ahkab import new_ac, run
from ahkab.circuit import Circuit
from ahkab.plotting import plot_results # calls matplotlib for you
import numpy as np

# Define the circuit
cir = Circuit('Butterworth 1kHz band-pass filter')
cir.add_vsource('V1', 'n1', cir.gnd, dc_value=0., ac_value=1.)
cir.add_resistor('R1', 'n1', 'n2', 50.)
cir.add_inductor('L1', 'n2', 'n3', 0.245894)
cir.add_capacitor('C1', 'n3', 'n4', 1.03013e-07)
cir.add_inductor('L2', 'n4', cir.gnd, 9.83652e-05)
cir.add_capacitor('C2', 'n4', cir.gnd, 0.000257513)
cir.add_inductor('L3', 'n4', 'n5', 0.795775)
cir.add_capacitor('C3', 'n5', 'n6', 3.1831e-08)
cir.add_inductor('L4', 'n6', cir.gnd, 9.83652e-05)
cir.add_capacitor('C4', 'n6', cir.gnd, 0.000257513)
cir.add_capacitor('C5', 'n7', 'n8', 1.03013e-07)
cir.add_inductor('L5', 'n6', 'n7', 0.245894)
cir.add_resistor('R2', 'n8', cir.gnd, 50.)

# Define the analysis
ac1 = new_ac(2.*np.pi*.97e3, 2.*np.pi*1.03e3, 1e2, x0=None)

# run it
res = run(cir, ac1)

# plot the results
plot_results('5th order 1kHz Butterworth filter', [('|Vn8|',"")], res['ac'],
             outfilename='bpf_transfer_fn.png')
```

Here is the full error:

```
Traceback (most recent call last):
  File "tlsim\main.py", line 30, in <module>
    outfilename='bpf_transfer_fn.png')
  File "C:\Users\usr\AppData\Local\Programs\Python\Python37-32\lib\site-packages\ahkab\plotting.py", line 235, in plot_results
    pylab.hold(True)
AttributeError: module 'pylab' has no attribute 'hold'
```

Sys:
 - Win10
 - Python 3.7.1

installation:

  python -m pip install ahkab[PLOT]

raczben, 2019-06-12:
I found in a [matplotlib issue](https://github.com/matplotlib/matplotlib/issues/12337/#issuecomment-425628982):

> The hold functionality was deprecated in 2.1, and intentionally removed in 3.0.

raczben, 2019-06-12:
For fast fix I suggest to add this restriction into setup.py:

Current:
```
extras_require = {'PLOT':['matplotlib>=1.1.1']},
````

Suggestion:
```
extras_require = {'PLOT':['matplotlib>=1.1.1,<3.0']},
````

RaphaelMaschinsen, 2023-07-20:
I have the same problem, the example code is not running.