ahkab/ahkab #75: Help : Modeling a circuit with parallel components

subash774, 2020-07-15:
Apologies for my incompetency, I am completely illiterate when it comes to electrical engineering. I've the circuit diagram which I want to model using python. 

Circuit diagram attached:
[image]

I've looked at the examples but none seems to match what I have, can anyone help me with modeling this please? How would parallel flows work?

Thanks in advance!

closed 2020-07-19 by subash774

ggventurini, 2020-07-19:
Dear Subash,
I am GIuseppe Venturini’s mother. Thanks a lot for writing.

Giuseppe passed away in September 2015 after a long illness.
He was very fond of Ahkab and enjoyed talking to other people interested in
electronics like he was. He worked at CERN and was doing a PHD at EPFL
He would have been happy to answer you.

Have you found a solution to your technical problem?
Otherwise, I’ll write to a friend of Giuseppe who helps with Ahkab and who
can surely answer your questions.

I wish you all the best
Lucia

subash774, 2020-07-19:
Hi Lucia,
Oh, I'm sincerely sorry to hear that. Thanks for replying. I've managed to solve my problem, thanks. 
May his soul rest in peace.